RobotWebTools/roslibjs #345: Documentation on how to make roslibjs work with bson only mode on rosbridge

VorpalBlade, 2019-10-04:
I'm interested in running with BSON only mode on the rosbridge side. However I can't figure out how to make this work, as I get errors like this (and no data passed):
```
2019-10-04 16:16:52+0200 [-] [ERROR] [1570198612.968579]: [Client 0] Exception in deserialization of BSON
```
I have added the bson module (I'm using webpack currently to generate the client side javascript, since the ready made browser bundle file did not work for bson).

```
const bson = require('bson');
const roslib = require('roslib');
```

It does not seem like roslib picks this up. Nor does it help modifying roslib to have it as a hard dependency in SocketAdapter.js. Any advice on how to get this working would be useful.

Rayman, 2019-10-04:
How do you know "roslib did not pick this up". What error did you get?

Did you get the following message? `throw 'Cannot process BSON encoded message without BSON header.';`

VorpalBlade, 2019-10-05:
No, I did not get that message. What happened is that I got a message that it managed to connect the web socket (from `ros.on('connection', ...`). Then nothing (except the errors on the server side from rosbridge_server as described above). If I disable bson_only_mode subscriptions go through and everything works.

This was the behaviour both with and without the `bson = require('bson')` line or not. As well as when I replaced the optional check in SocketAdapter.js with a hard dependency on bson.

trusktr, 2021-09-16:
@Rayman What he was talking about is that the `SocketAdapter.js` file looks for an undocumented global variable `bson` here:

https://github.com/RobotWebTools/roslibjs/blob/a07fa1d7ef42eac9373ab4925d1ecf17966778fc/src/core/SocketAdapter.js#L15-L17

If that variable exists (presumably from the `bson` package, one would assume, as it is not documented), then `SocketAdapter` will use it.

I'm catching errors like `'bson' is not defined` now that I'm running ESLint on the source while working on https://github.com/RobotWebTools/roslibjs/pull/475.

---

@VorpalBlade What you would have to do it set `bson` _as a global_ in Node.js:

```js
global.bson = require('bson');
```

The version you wrote using `const` does not create a global variable, therefore `roslib` does not see it.

trusktr, 2021-09-18:
Ah, I need to add a shim file for `bson` in #475. Documenting it would be nice.