RobotWebTools/roslibjs #352: roslibjs logging functions missing?

vashmata, 2020-02-07:
My team was looking into how to use ROS logging (loginfo, logerr, logwarn) in roslibjs for our project but it doesn't look like this functionality exists right now. If it does exist, how can I use it? Otherwise, I think it would be good to have.

Rayman, 2020-02-10:
On the browser side, I always use `console.log` because that is very well integrated in the chrome debugger (viewing/filtering)

If you want, you could create a publisher to the `rosout` topic, but this is not (yet) implemented in roslibjs.

vashmata, 2020-02-10:
I suppose that works, but to correctly publish to /rosout I would need to generate the seq, header and other data in the message.

vashmata, 2020-02-15:
I've made significant progress in writing functions to behave like the rospy log functions. Some things are still missing but I'm hoping to make a PR with my code at some point.

MatthijsBurgh, 2021-06-13:
@vashmata still planning to open a PR with logging functions?

vashmata, 2021-06-13:
> @vashmata still planning to open a PR with logging functions?

Thanks for the reminder, last year I actually did finish the code on our end. I'd like to make a PR, but where do I put the code? It's a script that has to be loaded after the ROSLIB.ros() object is created and ROS connection has been established. Not sure where in your file/folder structure it belongs. I created a little example webpage with buttons that demos the functionality too.

vashmata, 2021-06-13:
You can take a look at Logging.html and ros_log.js over here to get an idea:
https://github.com/vashmata/roslibjs/tree/ros_logging/examples

vashmata, 2022-01-13:
Made a PR to get some eyes on it, but the [CI test for `noetic` failed](https://github.com/RobotWebTools/roslibjs/runs/4772015126?check_suite_focus=true#step:5:477), not sure why. Here's the error:
```
Firefox 95.0 (Ubuntu 0.0.0) TF2 Republisher Example tf republisher FAILED
	Timeout of 2000ms exceeded. For async tests and hooks, ensure "done()" is called; if returning a Promise, ensure it resolves.
```
I don't know what this has to do with my repo, firstly because it's referencing a different example, and secondly because I'm not super familiar with async and whatnot and I don't think I even used that in my files.