RobotWebTools/roslibjs #360: Problem getting Publish to work

CycleMark, 2020-04-17:
Hi,

I'm trying to copy the example from the examples code to simply send/publish a string message. For some reason nothing seems to be being sent (I know the connection is up as I'm receiving all the data from the host ROS system). 

Is there anything I can check?

Example Code:

startstopclick = function(event) {      
      // Method to send command to Robot to start/stop sentry
      var robotCommand = new ROSLIB.Topic({
        ros : ros,
        name : '/robotCommand',
        messageType : 'std_msgs/String'
           });
      
      var mxnetSentryCommand = new ROSLIB.Message(
       "GO"
      );
      
      robotCommand.publish(mxnetSentryCommand);
    }

Kind Regards

Mark

Rayman, 2020-04-17:
Is there an error you get in the console?

CycleMark, 2020-04-18:
Hey - no nothing. I've break pointed it - check through and no exceptions/errors etc are being thrown.

On the server side I've checked with rostopic echo - and nothing it being received there either.

Cheers

Mark

Rayman, 2020-04-19:
Strange. Can you check in the browser the developer tools (F12), Network and then WS (websocket). Here you can see the messages that are send between rosbridge and the browser. 

Are there any messages send? If not the problem is in the javascript code.

CycleMark, 2020-04-20:
Checking that debug I can't see anything. That said there is no incoming traffic either - which is working (as I can see that)

Thanks
Mark

mvollrath, 2020-04-20:
The problem is you are not forming the message properly.

There is probably a message in the rosbridge log that looks like:

```
2020-04-20 12:43:22+0000 [-] [ERROR] [1587386602.531928]: [Client 7] [id: publish:/foo:2] publish: Message type std_msgs/String does not have a field 1
```

This is wrong:

```javascript
var mxnetSentryCommand = new ROSLIB.Message(
 "GO"
);
```

You have to specify which field in the message the string "GO" is in.  Since there is only one field in std_msgs/String, this is right:

```javascript
var mxnetSentryCommand = new ROSLIB.Message({
 data: "GO"
});
```

closed 2020-04-20 by mvollrath

CycleMark, 2020-04-20:
Perfect thanks! 

Not sure why there was no error message :-/

Thanks Mark