RobotWebTools/roslibjs #364: getParam requires rosapi?

130s, 2020-05-14:
I posted [answers.ros.org#q352354](https://answers.ros.org/question/352354/roslibjsgetparam-returns-service-rosapiget_param_names-does-not-exist/) but getting convinced about root of issue so posting here too. Apologies for duplication.

Experimenting at my team, running `rosapi` node seems to be a must in order for roslibjs to `getParam` ROS parameters. I haven't been able to find the documentation for that though. So,

- Is rosapi node running a requirement for getParam to work?
- If so, why not rosbridge_server start rosapi internally?
- Where's the right place to document this (I'm not js dev so I'm not familliar with RWT codebase)?

mvollrath, 2020-05-15:
> * Is rosapi node running a requirement for getParam to work?

Yes.

> * If so, why not rosbridge_server start rosapi internally?

rosapi is an optional extension of rosbridge.  For example, most roslib projects I've worked on have not needed rosapi because they do not access ROS parameters.

> * Where's the right place to document this (I'm not js dev so I'm not familliar with RWT codebase)?

Making a note in the docstrings of the roslib param methods seems appropriate.

130s, 2020-05-18:
>>  -     Is rosapi node running a requirement for getParam to work?
>
> Yes.

Thank you for the confirmation. Since it was the wiki.ros.org that we relied upon (and ultimately got confused), I added some info about `rosapi` requirement http://wiki.ros.org/action/recall/roslibjs/Tutorials/BasicRosFunctionality?action=recall&rev=17.

130s, 2020-05-18:
> -        If so, why not rosbridge_server start rosapi internally?
>
> rosapi is an optional extension of rosbridge. For example, most roslib projects I've worked on have not needed rosapi because they do not access ROS parameters.

In ROS (or I should say ROS1 because I don't know about ROS2), the basic way to start ROS system is by `roscore`, which also starts ROS Parameter server. So it's fair to say the developers with C++ and Python language clients assume parameter server can be accessed without starting any additional nodes.

I do not know much about the requirement that the rosbridge had/has considered, but if very few people have complained about ROS Parameter not being accessible without rosapi, then there might not be much need.

Either way, I'm just sharing my thoughts without strong opinion. Closing as minimal improvement is done IMO.

closed 2020-05-18 by 130s