RobotWebTools/roslibjs #365: getting weird "missing uws" issue

flynneva, 2020-05-19:
so i've got this project that uses roslibjs. everythings been great, but once I add docz to help with documentation and run docz:dev, which builds everything before starting a dev server hosting the docs site i get this weird error saying im missing uws. 

```
ERROR #98124  WEBPACK
Generating SSR bundle failed
Can't resolve 'uws' in '/home/flynn/code/web/ros/react-ros/node_modules/roslib/node_modules/engine.io/lib'
If you're trying to use a package make sure that 'uws' is installed. If you're trying to use a local file make sure that the path is correct.
File: ../node_modules/roslib/node_modules/engine.io/lib/server.js
```

it was exactly what was described on this [issue over on socketio](https://github.com/socketio/socket.io/issues/3342) and so I tried their solution and it solved my issue as well.

anyone else getting a similar error, or just me?

Rayman, 2020-05-20:
roslibjs doesn't depend on uws, so what does it have to do with this project?

flynneva, 2020-05-20:
its caused by a dependency of this project, socket.io. I was just asking if anyone had experienced a similar issue. apologies.

closed 2020-05-20 by flynneva

Rayman, 2020-05-20:
Ah OK, you are right. I see roslibjs uses `socket.io` at version 2.2.0. Does that give this error and if yes, should we upgrade the version?