RobotWebTools/roslibjs #371: Report service errors from "op": "status"

kaikoehn, 2020-06-30:
I'm using [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge/) as a web-bridge and noticed that when doing a service call which caused a `TypeError` on the webbridge the error was not reported in any of the callbacks (callback, failedCallback) of the client.

Client library call:
`service.callService(new ROSLIB.ServiceRequest(args), handleResponse, handleError)`

WS Request:
```json
{
  "op": "call_service",
  "id": "call_service:coco/v1/stop_mapping:2",
  "service": "coco/v1/stop_mapping",
  "type": "api_v1_interfaces/srv/StopMapping",
  "args": {
    "foo": 1
  }
}
```

WS Response:
```json
{
  "op": "status",
  "level": "error",
  "msg": "call_service: TypeError: Invalid argument: generate_localization_map in StopMapping_Request",
  "id": "call_service:coco/v1/stop_mapping:2"
}
```

Are there plans to to propagate this kind of status with `roslibjs` as well? Or is this some specifics of `ros2-web-bridge`?

I could try to make a PR but I'm not sure if this is part of the standard that should be supported or not.

Rayman, 2020-06-30:
`callService` has to following signature, so on error I would expect `failedCallback` to be called. You say this is not happening?
```js
Service.prototype.callService = function(request, callback, failedCallback) {
```

kaikoehn, 2020-06-30:
Correct - neither of the callbacks were called for me. I was using the npm package / v.1.1.0.

Rayman, 2020-07-01:
A PR to fix that would be appreciated :smile: