RobotWebTools/roslibjs #372: How to subscribe a ros topic which user defined self?

herogan2017, 2020-07-21:
var listener = new ROSLIB.Topic({
    ros : ros,
    name : '/slam/apa_location',
    messageType : 'msgs/apa_location'
  });

I hava a messageType defined by myself, not standard type. 
How to suscribe it using roslibjs?

thanks.

Rayman, 2020-07-21:
The same way as subscribing to any other message.

herogan2017, 2020-07-21:
> The same way as subscribing to any other message.

Thank you. I solve it.

closed 2020-07-21 by herogan2017

Surtr07, 2022-10-26:
Can you explain how to do it exactly? I want to use simple.html to listen a topic which msg defined by me.

MatthijsBurgh, 2022-10-26:
I think the simple example is very straightforward. Change the `messageType` value to your message name. And change the object you provide to `ROSLIB.Message`.

Surtr07, 2022-10-26:
> I think the simple example is very straightforward. Change the `messageType` value to your message name. And change the object you provide to `ROSLIB.Message`.

What do you mean change the  'ROSLIB.Message'?Isn`t the 'ROSLIB.Message' the content for publish a topic?

  var twist = new ROSLIB.Message({
    linear : {
      x : 0.1,
      y : 0.2,
      z : 0.3
    },
    angular : {
      x : -0.1,
      y : -0.2,
      z : -0.3
    }
  });

  cmdVel.publish(twist);

  var listener = new ROSLIB.Topic({
    ros : ros,
    name : '/obstacles_filter',
    messageType : 'common_msgs/DetectedObjectArray'
  });

  listener.subscribe(function(message) {
    console.log('Received message on ' + listener.name + ': ' + message.data);

Anything else should I change?

MatthijsBurgh, 2022-10-26:
Sorry, Indeed you need to define a new message for publishing. For subscribing, the message is provided by `ROSLIB`, so the message has the correct structure.

Surtr07, 2022-10-26:
> 

do you mean I can`t use a listener to show a topic which msg defined by myself? or how can I add a customized msg in ROSLIB?

MatthijsBurgh, 2022-10-26:
ROSLIB, will do this for you. No need to register this custom msg.