RobotWebTools/roslibjs #376: Subscribe to any type

mirkow, 2020-10-02:
Hi,

I would like to subscribe to a topic without knowing its type. And then use reflection to display the data. Is this possible with roslibjs?
If not, is there a way to first query the type of the topic and the use this to subscribe to that topic?

Thanks
Mirko

jorgenfb, 2020-11-02:
Hi! I'm assuming you are running `rosbridge_suite` as your backend. `rosbridge_suite` includes `rosapi` to do its work. `rosapi` provides a lot of useful services that would probably suite your needs, including:

`/rosapi/topic_type` to lookup the type for a given topic
`/rosapi/topics` get all topics and their type

You can find all of them with `rosservice list | grep rosapi`

MatthijsBurgh, 2021-06-13:
Solution provided

closed 2021-06-13 by MatthijsBurgh