RobotWebTools/roslibjs #377: NODE-RED with ROS, by roslib, TypeError: util.debug is not a function

wei1224hf, 2020-10-14:
my node-red is V 1.1.3
my ROS is Kinetic
my node-red-contrib-ros is 1.0.8

when I want to communicate ROS with Node-Red, there is a error on Node-Red:
`TypeError: util.debug is not a function`

And I solve it by change the code in roslib:

roslib\src\node\RosTCP.js
comment the line 13 and line 14:

`    //util.debug('ROSLib uses utf8 encoding by default.' +
    //  'It would be more efficent to use ascii (if possible)');`

Rayman, 2020-10-14:
When I look at [src/node/RosTCP.js](https://github.com/RobotWebTools/roslibjs/blob/6c17327cae14ca0c76ca5f71de5661279207219c/src/node/RosTCP.js#L16) I dont' see any reference to `util`. Are you sure the bug is in this package. It could be an issue in the integration of `node-red-contrib-ros`

wei1224hf, 2020-10-18:
I install the RosTCP.js by :
sudo apt-get install ros-kinetic-rosbridge-suite
in this way ,there is util.debug in RosTCP.js

closed 2020-10-18 by wei1224hf