RobotWebTools/roslibjs #380: Swift/iOS port?

christophomos, 2020-11-11:
I posted this question on https://groups.google.com/g/robot-web-tools, but Jihoon suggested I open an issue here:

I'm looking to contribute my iOS skills to an open source project, and I'm wondering if an iOS port of roslibjs would be valuable. Being able to directly control ROS devices using mobile devices via websockets and perhaps later with BLE could open up interesting possibilities.

An example analogous use case is the Donkey Car mobile app:
http://docs.donkeycar.com/guide/mobile_app/

trusktr, 2021-09-18:
We can already run roslibjs in a webview in Android or iOS. The webview would have the WebSocket API with which roslib communicates with.

Or, in other words, roslibjs is built for web, and the web runs on all devices, therefore roslibjs already can run on any device where a web browser or webview exists.

If you want to specifically port to another language, that could be a fun project. It would be more valuable for ROS in general, not necessarily valuable to roslibjs which is meant for web. F.e. you could make something like https://github.com/RobotSwiftTools/roslib.swift.

christophomos, 2021-09-18:
Makes sense. At any rate, I'm working on a VR project instead.

Rayman, 2021-09-28:
After some quick googling I found 2 ROS2 projects that might interest you:
- https://github.com/younata/rclSwift
- https://github.com/atyshka/ros2_swift

closed 2023-02-05 by MatthijsBurgh