RobotWebTools/roslibjs #395: Issue with ROS2 bridges and roslibjs services/actions

AndrewJSchoen, 2021-01-28:
Hi all. I have been posting a number of issues across the org lately trying to decipher exactly where an issue is occurring with ROS2 versions of many of these libraries. I think, based on some of the errors and feedback in the following issues that it come down to the roslibjs library. 

Existing Issues:
[rosbridge_suite:552](https://github.com/RobotWebTools/rosbridge_suite/issues/552)
[ros2-web-bridge:#171](https://github.com/RobotWebTools/ros2-web-bridge/issues/171)

Basically, it seems that when roslibjs attempts to create anything other than a publisher/subscriber connection, it looks for the wrong message type (i.e., the ROS1 version instead of the ROS2 version):

(from the rosbridge_suite issue):
```
[ERROR] [1611770934.850646234] [rosbridge_websocket]: [Client 0] [id: advertise:/tf2_web_republisher/goal:55] advertise: Unable to import msg class TFSubscriptionActionGoal from package tf2_web_republisher. Caused by module 'tf2_web_republisher.msg' has no attribute 'TFSubscriptionActionGoal'
[ERROR] [1611770934.932197810] [rosbridge_websocket]: [Client 0] [id: subscribe:/tf2_web_republisher/feedback:57] subscribe: Unable to import msg class TFSubscriptionActionFeedback from package tf2_web_republisher. Caused by module 'tf2_web_republisher.msg' has no attribute 'TFSubscriptionActionFeedback'
```

For exhaustion of links, it seems like there is some discussion of this parity issue: [ros2-web-bridge:148](https://github.com/RobotWebTools/ros2-web-bridge/issues/148)

Is there any interest in remedying this?

In the meantime, we might be able to co-opt the fact that roslibjs just ends up expecting "topics" regardless of it being a topic, service, or action (from what I see). Might I just be able to create a bunch of messages and topics that mirror what roslibjs is expecting? Am I right in this understanding?

mypsybot, 2022-06-27:
ROS 1 uses topics for action but ROS 2 uses a combination of services and topics. roslibjs uses a topic implementation ie, ROS 1. I am also looking for a fix but am unable to find it. I currently use a ROS 2 action client with the subscriber in python which listens to roslibjs and sends the goal.

nPellejero, 2022-08-18:
Hi guys, any news on this matter? we have been developing a bunch of actions and when trying to connect to the web interface we run into this problem.. any advances? if not, @mypsybot  could you give some more detail of your workaround solution please?

AltinayGrass, 2022-09-30:
Hi all,
Same problem here. I am testing fibonacci action server. Action server is  running on shell. Client side on web page.

I am getting below messages.
`[INFO] [1664526743.377902751] [rosbridge_websocket]: [Client e398c734-70dd-4a8e-9f1b-3619155e37c0] Subscribed to /fibonacci/status
[ERROR] [1664526743.381021699] [rosbridge_websocket]: [Client e398c734-70dd-4a8e-9f1b-3619155e37c0] [id: subscribe:/fibonacci/feedback:4] subscribe: Unable to import action_tutorials_interfaces.msg from package action_tutorials_interfaces. Caused by: No module named 'action_tutorials_interfaces.msg'
[ERROR] [1664526743.384244908] [rosbridge_websocket]: [Client e398c734-70dd-4a8e-9f1b-3619155e37c0] [id: subscribe:/fibonacci/result:5] subscribe: Unable to import action_tutorials_interfaces.msg from package action_tutorials_interfaces. Caused by: No module named 'action_tutorials_interfaces.msg'
[ERROR] [1664526743.389325059] [rosbridge_websocket]: [Client e398c734-70dd-4a8e-9f1b-3619155e37c0] [id: publish:/fibonacci/goal:6] publish: Cannot infer topic type for topic /fibonacci/goal as it is not yet advertised
`

MatthijsBurgh, 2022-11-22:
I will repeat myself: "I don't know anything about integration/compatibility of ROS2. Maybe https://github.com/orgs/RobotWebTools/teams/ros2-0 can help you out."

xkaraman, 2022-11-22:
Sure, i am just asking whom ever is familiar since it seems it got lost. :)

sea-bass, 2023-12-20:
We recently merged official ROS 2 action support in both rosbridge_suite and this repo:

* https://github.com/RobotWebTools/rosbridge_suite/pull/886
* https://github.com/RobotWebTools/roslibjs/pull/645

There are tests and examples with these changes.

If this is still applicable for people in this thread, I hope you get to try these out!

closed 2023-12-20 by sea-bass

RafaaZahra, 2024-01-15:
I have same error , and i try to update ros_bridge for ros2 Humble , 
# code #

var  ActionClient = new ROSLIB.ActionClient ({ 
       ros : ros , 
       serverName : '/move_robot',
       actionName : 'robot_interfaces/action/MoveRobot'  //this action interface from ros2 server 
}); 

I am getting below messages:
[ERROR] [1705349916.420697849] [rosbridge_websocket]: [Client dcc57d31-e881-4795-bb12-4763dbff8a61] [id: subscribe:/move_robot/result:5] subscribe: Unable to import robot_interfaces.msg from package robot_interfaces. Caused by: No module named 'robot_interfaces.msg'
[rosbridge_websocket-1] [ERROR] [1705349916.420937187] [rosbridge_websocket]: [Client dcc57d31-e881-4795-bb12-4763dbff8a61] [id: publish:/move_robot:6] publish: Cannot infer topic type for topic /move_robot as it is not yet advertised