RobotWebTools/roslibjs #396: Maintainers, please respond

MatthijsBurgh, 2021-02-02:
There has not been any activity in this repo since midway december 2020. I have created 2 PRs, there are many dependabot PRs open. But there seems to be no activity from the maintainers.

I don't expect next day responses, but no activity for 1,5 month is very disappointing and discouraging to keep contributing. So please stay a little bit active here. Thanks!

mvollrath, 2021-02-02:
I do not have bandwidth to maintain this presently, @jihoonl may step in to add maintainers.

MatthijsBurgh, 2021-02-11:
@jihoonl I would be happy to help out.

MatthijsBurgh, 2021-02-24:
ping ;)

MatthijsBurgh, 2021-03-08:
@jihoonl @mvollrath a little bit less friendly ping

jihoonl, 2021-03-08:
@MatthijsBurgh sorry for noticing the issue this late. I don't have a bandwidth to handle the PRs at the moment. I will try my best to review the pending PR within March. Please understand it takes time. I have been away with github, so I would need to take some time to catch up what the GH action is.

Again, sorry for late response.

MatthijsBurgh, 2021-03-30:
@jihoonl But how about adding me as a maintainers? I don't plan to push on big changes, but at least I can take care of the dependabot PRs and similar small changes.

jihoonl, 2021-04-04:
Hi @MatthijsBurgh, sorry for late response again. It took me a long time to accept the fact that I am not able to follow up the new features of github and javascript trend. Probably, it is right to give the maintainership to the right people to keep the library maintained and moving forward.

I have invited you as a mainainer for roslibjs, ros2djs, and ros3djs. Thank you for stepping up and willing to help. Please let me know if you need more access. And send me e-mail to jihoonlee.in at gmail if I am not responding in a github issue. I will try my best to help out.

jihoonl, 2021-04-04:
FYI, @mvollrath @J-Rojas @minggangw

closed 2021-04-05 by MatthijsBurgh