RobotWebTools/roslibjs #400: Topic: throttle_rate and queue_length not supported?

jtrautmann, 2021-02-24:
I'm using the `throttle_rate` and `queue_length` like this:

```
var image_topic = new ROSLIB.Topic({
    ros : ros,
    name : image_topic_name,
    messageType : 'sensor_msgs/Image',
    queue_length : 1,
    throttle_rate : 5000
});

image_topic.subscribe(function(message) {
    // ...
});
```

I would expect to receive an image message each 5 seconds that is up to date. But instead between two messages I only have around 200 milliseconds. Changing the throttle_rate doesn't seem to change anything. The images are also pretty out of date, so it doesn't seem the queue is working.
Are the `throttle_rate` and `queue_length` options really supported?

jtrautmann, 2021-02-24:
This seems to be an issue with ros2-web-bridge and not with roslibjs.

closed 2021-02-24 by jtrautmann

ManuelZ, 2021-04-05:
Hi @jtrautmann Do you have more details on this? Where did you find out that this was a ros2-web-bridge issue?