RobotWebTools/roslibjs #419: Is ROSLIB.Param compatible with ROS2?

ManuelZ, 2021-04-08:
In the examples that I've seen the name parameter of ROSLIB.Param never considers the node name. Is it possible to get/set parameters with ROS2?

trusktr, 2021-09-18:
@ManuelZ Hello! If you can provide an example of any limitation you hit, please do. I've used roslib with ROS2, but not the params.

jastion, 2022-05-03:
@trusktr @ManuelZ 
Just wanted to leave a comment for future users interested in this.
I am able to get/set ROS2 parameters using roslibjs
System:
Ubuntu 20.04LTS running Foxy
roslibjs on ros2 branch

Followed the example shown in simple.html
format for naming for ros parameter in code as
name : "/{NODE_NAME}:{PARAM_NAME}", 
ie: name: "/simple_node:param",

mypsybot, 2022-06-21:
For me, I am using roslibjs on ROS2. I have not explored it entirely but the basics seem to work okay.
Ubuntu 20.04 running Foxy

sea-bass, 2023-12-20:
Yes, it should be. See here for an example: https://github.com/RobotWebTools/roslibjs/blob/develop/examples/ros2_simple.html#L129

closed 2023-12-20 by sea-bass