RobotWebTools/roslibjs #433: Publish/update existing npm release

surajhpatil, 2021-05-05:
It seems @mvollrath with #353  released new package to npm registery with version `1.1.0`.

However, it seems like `package.json` version from npm released `roslib-1.1.0` is outdated.
Because of this, npm audit now gives vulnerability issues to roslib for `socket.io` and `xmldom`.

An example of outdated package:
Current `socket.io` in npm package is `2.2.0` where as in `develop` it's `4.0.1`.
Current `xmldom` in npm package is `^0.1.19` where as in `develop` it's `^0.6.0`.

To test:
Try downloading the npm published roslib 1.1.0 using this URL: https://registry.npmjs.org/roslib/-/roslib-1.1.0.tgz
Extract and inspect the `package.json` and compare it with projects [current package.json](https://github.com/RobotWebTools/roslibjs/blob/develop/package.json)

This could be resolved by updating existing npm release or though a new release.

MatthijsBurgh, 2021-05-05:
I wouldn't call this outdated. This is just the process of developing and releasing.

There is no need to compare the npm released version and  the `develop` version.

The npm released version matches the [tagged version](https://github.com/RobotWebTools/roslibjs/releases/tag/1.1.0).

A new version will be released in a few weeks. If you need the current `develop` version. NPM also support git dependencies.

surajhpatil, 2021-05-05:
> A new version will be released in a few weeks. If you need the current `develop` version. NPM also support git dependencies.

Sounds good! Closing this issue :+1:

closed 2021-05-05 by surajhpatil