RobotWebTools/roslibjs #452: Demos on website don't work

saulzar, 2021-06-17:
For example if I click on Demos -> Fetch URDF I get this in the console and only showing a grid. The other demos have similar errors.

```
Failed to load resource: the server responded with a status of 403 ()
static.robotwebtools.org/ColladaAnimationCompress/current/ColladaLoader2.js:1 Failed to load resource: the server responded with a status of 403 ()
ros3d.min.js:1 THREE.WebGLRenderer 89
roslib.min.js:1 WebSocket connection to 'ws://demo.robotwebtools.org:9090/' failed: 
d.connect @ roslib.min.js:1
roslib.min.js:1 Uncaught Error: Uncaught, unspecified 'error' event.
    at d.h.emit (roslib.min.js:1)
    at WebSocket.onerror (roslib.min.js:2)
```

trusktr, 2021-09-18:
Seems the backend is (or parts of it are) down.

vkuehn, 2022-02-24:
atill an issue

MatthijsBurgh, 2022-02-24:
Don't expect them to work anymore. You can run them locally.

closed 2022-02-24 by MatthijsBurgh