RobotWebTools/roslibjs #455: The callback function is not executed after subscribing  to a topic

luyao514, 2021-06-24:


Rayman, 2021-06-24:
Which callback? Can you provide more details?

luyao514, 2021-06-24:
> Which callback? Can you provide more details?
The following code is invalid. The callback function is not executed.The MessageType here is custom
`        var listener = new ROSLIB.Topic({
            ros: ros,
            name: '/detection/hook_ladle_fusion',
            messageType: 'api_msgs/HookLadle'
        });

        listener.subscribe(function (message) {
            console.log('Received message on ' + listener.name + ': ' + message.data);
            // listener.unsubscribe();
        });
`

trusktr, 2021-09-18:
Is the message being emitted? Is your client connected to ROS? etc?

I have only had this issue in the past if I wrote the wrong topic name by mistake, or tried to connect to the wrong ROS server (or no server at all), etc.

luyao514, 2021-09-18:
Thank you very much! I have solved the problem, the previous issue was due to my messageType is wrong

closed 2021-09-18 by MatthijsBurgh