RobotWebTools/roslibjs #497: How to get a web page to automatically reconnect after losing connection to ROSBridge

iandanielsooknanan, 2021-11-29:
**Description**
Hello, we are developing a webpage to control our robot using ROS and roslibjs. The webpage is hosted on the robot's computer while a tablet will access the webpage for control.
We require the user of the webpage (on the tablet) to know if the connection between it and the robot's computer is lost.
We have achieved this by using the emitters for connection, error and close to print to screen or popup with the message. 
However, we also require the webpage to try to reconnect in the event that the connection was lost. 
This is what we were not able to achieve. 
Some solutions that failed:
Refreshing the page: We can't refresh the page because we need to keep some variables of the current session.
Placing another connection statement: 
`ros = new ROSLIB.Ros({
        url: "ws://" + robot_IP + ":9090"
    });`
inside of the close emitter but this failed as well.
Are there any suggestions on how to achieve this?

- Library Version: latest
- ROS Version: Melodic 
- Platform / OS: Ubuntu 18.04

**Steps To Reproduce**

The simple.html file in the examples folder is a similar to what we have. 
Start up ROSBridge and then open the page.
The console will report the connection.
Close ROSBridge.
The console will report the disconneciton.
Start up ROSBridge.
No activity in the console.

**Expected Behavior**
Webpage recconnects and the console notifies the user.

**Actual Behavior**
No reconnecition happens.

MatthijsBurgh, 2021-11-29:
This is not part of roslib. Though I have created a small library myself a while back. https://GitHub.com/tue-robotics/auto-ros

harshanirudh, 2022-07-14:
can this be implemented as part of this package?