RobotWebTools/roslibjs #500: New release

MatthijsBurgh, 2021-12-21:
I am planning to release a new version.

I got some release instruction from @jihoonl, https://github.com/RobotWebTools/roslibjs/blob/release_process_document/RELEASE.md

Are there any issues that people want to have fixed before a new release. Please comment. I am aiming for a new release in the first week of January.

I don't think we have any breaking changes, so it will be a minor version bump. (https://github.com/RobotWebTools/roslibjs/compare/1.1.0...develop)

MatthijsBurgh, 2021-12-21:
@RobotWebTools/roslibjs-maintainers @Rayman Please let me know what issues need to be fixed before a new release. Or when you have any comments on the release procedure.

Rayman, 2021-12-21:
Due to the huge amount of "bump" commits I can't easiliy review what has changed...

MatthijsBurgh, 2021-12-21:
I think if you check the changes per file, you get a good overview of the changes.

jihoonl, 2021-12-22:
I think it would be better to deprecate CDN and keep only NPM versions. The CDN maintainer has been away for quite a long time.

jorgenfb, 2022-01-05:
It would be great if #489 could be included. It prevents a leak in TFClient and should be fairly easy to include.

MatthijsBurgh, 2022-01-08:
`1.2.0` has been released

closed 2022-01-08 by MatthijsBurgh