RobotWebTools/roslibjs #501: Not able to init websocket connection: ws does not work in the browser

akshay-nm, 2021-12-24:
**Description**
I am trying to setup a web interface using svelte and this package with a turtlebot3 simulation on ros2 galactic. 
I have used `npm install roslib` to set up this npm module in my project.

This is what I have in my client-side script (JS): 
```
ros = new ROSLIB.Ros({
  url: 'ws://localhost:9090'
})
```
I have the simulation up and running with rosbridge_suite and rosbridge_server is up on port 9090 of localhost.

- Library Version: "roslib": "^1.1.0"
- ROS Version: galactic
- Platform / OS: wsl2 (ubuntu 20.04)

**Expected Behavior**
I should be able to set up a WebSocket connection.

**Actual Behavior**
When I open up my web interface I get this error: 
```
Uncaught (in promise) Error: ws does not work in the browser. Browser clients must use the native WebSocket object
    at new module.exports (browser.js:4)
    at Ros.connect (Ros.js:75)
    at new Ros (Ros.js:53)
    at getRosConnection (ros-connection.js?t=1640387808327:11)
    at __layout.svelte? [sm]:9
    at run (index.mjs:18)
    at Array.map (<anonymous>)
    at index.mjs:1749
    at flush (index.mjs:1057)
    at init (index.mjs:1841)
```

desstiony, 2022-06-10:
ROS Version: galactic is ros2. you sholud use [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge)

MatthijsBurgh, 2022-06-14:
`rosbridge_suite` has a ROS2 version. `ros2-web-bridge` is not actively maintained...

EzraBrooks, 2025-11-03:
This is definitely fixed on `develop` at this point, as the logic for selecting WebSocket or ws is much better now.