RobotWebTools/roslibjs #504: JSDoc is 404 now.

eastpiger, 2021-12-28:
**Description**

JSDoc is 404 now.

- Library Version:
- ROS Version:
- Platform / OS:

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

MatthijsBurgh, 2021-12-28:
Caused by commit https://github.com/RobotWebTools/robotwebtools.github.io/commit/4f3523dae536c0832c24c21c539a5e0020987174 of  @defunctzombie

The docs should become available again online in the (nearish) future.

You can still generate the docs yourself by running `npm run doc`.

MatthijsBurgh, 2022-01-10:
I have uploaded the `1.2.0` docs. You can find them at http://robotwebtools.org/roslibjs/

closed 2022-01-10 by MatthijsBurgh

defunctzombie, 2022-01-10:
I don’t see any updates on the website repo - so how is this new
information hosted on the robotwebtools.org site?

I also think that domain is not the place to host docs for all the various
tools or libraries. Each library should host its own docs.

MatthijsBurgh, 2022-01-11:
@defunctzombie The docs are hosted in the `roslibjs` repo.

Each repo has its own webpage. The `organization.github.io` repo provides the mainpage. Then every repo in the organization can provide `organization.github.io/repo`. I don't know what happens when both `organization.github.io` has a subfolder of `repo_a` and `repo_a` itself provides a page.

So the `roslibjs` docs are now on the `gh-pages` branch of this repo.

defunctzombie, 2022-01-11:
@MatthijsBurgh wow - I had no idea this per-repo fall-through behavior existing! I thought once you setup the organization repo that would handle all requests. That's both interesting and surprising. TIL