RobotWebTools/roslibjs #546: timeout is not affecting ActionClient and its Goal

ravijo, 2022-05-13:
**The timeout is not affecting ActionClient and its Goal**

- Library Version: 1.0.1
    ```console
    $ console.log(ROSLIB.REVISION)
    1.0.1
    ```
- ROS Version: Noetic
- Platform / OS: Ubuntu 20.04.4 LTS

**Steps To Reproduce**
1. Run the following commands in the terminal:
    ```console 
    $ roscore
    $ rosrun actionlib_tutorials fibonacci_server
    $ roslaunch rosbridge_server rosbridge_websocket.launch
    ```
1. Open the following HTML file in browser and check the console:
    ```html
    <!DOCTYPE html>
    <html>
    <head>
      <meta charset="utf-8" />
      <script src="roslib.min.js"></script>
    
      <script>
        var ros = new ROSLIB.Ros({ url: 'ws://localhost:9090' });
        var timeout = 5; // 5 seconds
        var actionClient = new ROSLIB.ActionClient({
          ros: ros,
          serverName: '/fibonacci',
          actionName: 'actionlib_tutorials/FibonacciAction',
          timeout: timeout
        });
    
        var goal = new ROSLIB.Goal({
          actionClient: actionClient,
          goalMessage: { order: 20 }, // setting the order really high to check timeout
        });
    
        goal.on('feedback', function (feedback) {
          console.log('Feedback: ' + feedback.sequence);
        });
    
        goal.on('result', function (result) {
          console.log('Final Result: ' + result.sequence);
        });
    
        goal.on('timeout', function () {
          console.log('timeout while sending the goal');
        });
    
        actionClient.on('timeout', function () {
          console.log('timeout while connecting to the action server');
        });
    
        goal.send(timeout);
      </script>
    </head>
    
    <body>
      <h1>Fibonacci ActionClient Example</h1>
    </body>
    </html>
    ```

**Expected Behavior**
After 5 seconds the timeout for the goal should occur and it should print `timeout while connecting to the action server`

**Actual Behavior**
The timeout (for connecting the action client and sending the goal) occurs at the beginning. Below is  the output in the console:

```console
timeout while connecting to the action server
timeout while sending the goal
Feedback: 0,1,1
Feedback: 0,1,1,2
Feedback: 0,1,1,2,3
Feedback: 0,1,1,2,3,5
Feedback: 0,1,1,2,3,5,8
Feedback: 0,1,1,2,3,5,8,13
Feedback: 0,1,1,2,3,5,8,13,21
Feedback: 0,1,1,2,3,5,8,13,21,34
Feedback: 0,1,1,2,3,5,8,13,21,34,55
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377,610
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377,610,987
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377,610,987,1597
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377,610,987,1597,2584
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377,610,987,1597,2584,4181
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377,610,987,1597,2584,4181,6765
Feedback: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377,610,987,1597,2584,4181,6765,10946
Final Result: 0,1,1,2,3,5,8,13,21,34,55,89,144,233,377,610,987,1597,2584,4181,6765,10946
```

Rayman, 2022-05-13:
You should probably attach the timeout event to the action client (instead of the goal)

ravijo, 2022-05-13:
@Rayman 

Thank you very much for your response. I think I have attached the timeout event to both of them (client and goal). Can you please check once again?

Moreover, timeout has different meanings as mentioned below:
1. **Timeout attached on goal:** Timeout while sending the goal
    * For example, the action server is taking a long time doing some heavy computation
2. **TImeout attached on the client:** Timeout while connecting to the action server

Rayman, 2022-05-13:
Found it, the argument for timeout is in milliseconds. Try changing it to 5000 then your example works

ravijo, 2022-05-13:
Ahhhh... I should have a little extra. the timeout used to be [ros::Duration](https://docs.ros.org/en/api/rostime/html/classros_1_1Duration.html) is C++ which made me think that it is in seconds.

Anyways, I am closing this issue. Thank you very much!

closed 2022-05-13 by ravijo