RobotWebTools/roslibjs #548: Issue with roslibjs and Vite

kavidey, 2022-05-19:
I am using roslibjs with a project that uses Vite as the build system and am running into an issue when running the project in development.

- Library Version: 1.3.0
- ROS Version: Galactic
- Platform / OS: Ubuntu 20.04
- Vite Version: 2.9.9

**Steps To Reproduce**
Minimum reproducible project here: https://github.com/kavidey/vite-sample-roslib
Run `npm run dev` to get the error, or `npm run build` then `npm run preview` to see the project working properly.

**Expected Behavior**
In both dev and preview mode, importing roslibjs should work without any errors.

**Actual Behavior**
When running the code in dev mode, I get the error: `ReferenceError: Can't find variable: Buffer`.

That error traces back to `core/Ros.js` requiring `ws`. I think that the `ws` shim may not be working correctly when vite is running in dev mode and roslibjs thinks that its running in node so it tries to require `ws`, though I'm not sure.

MatthijsBurgh, 2022-05-20:
I have really no clue. I also have to say this is a bit out of the support scope. I have to say I have never experienced any issue with webpack-dev-server.

When you are able to provide a solution that requires to be implemented in the library, I am open to that. But I am not able to help you finding that solution.

desstiony, 2022-06-10:
Just replace import 'roslib/build/roslib' with import 'roslib/build/roslib

MatthijsBurgh, 2022-06-10:
@desstiony you suggest to replace it with the same....

desstiony, 2022-06-10:
Sorry, I made a typo.  you should replace  "import * as ROSLIB from 'roslib'" with 'roslib/build/roslib' in main.js file.

MatthijsBurgh, 2022-06-10:
So you mean replace `import * as ROSLIB from 'roslib'`  with  `import * as ROSLIB from 'roslib/builld/roslib'`?

desstiony, 2022-06-10:
Just `import 'roslib/build/roslib'`, you can try it.

MatthijsBurgh, 2022-06-10:
I am not using Vite. It wasn't clear to me. So I thought it might not be clear to other people.

desstiony, 2022-06-10:
I have tried it in the codespace of this [repository](https://github.com/kavidey/vite-sample-roslib) hope it can help you. @kavidey

Interpause, 2022-07-23:
I think this issue might be related to https://github.com/RobotWebTools/roslibjs/issues/501. I think a fix would be to replace node.js `ws` dependency with <https://www.npmjs.com/package/isomorphic-ws> instead.

loran-luo, 2024-04-10:
> Just replace import 'roslib/build/roslib' with import 'roslib/build/roslib'只需将 import 'roslib/build/roslib' 替换为 import 'roslib/build/roslib'

Is there any difference?

Ticket2Andromeda, 2025-06-03:
Posting here for future viewers. I was using vite and struggling with the same build error with vite and used import 'roslib/build/roslib' with success.