RobotWebTools/roslibjs #557: Support rosbridge Fragmentation

nathanmargaglio, 2022-08-01:
The [rosbridge protocol provides the ability to fragment messages](https://github.com/RobotWebTools/rosbridge_suite/blob/ros2/ROSBRIDGE_PROTOCOL.md#311-fragmentation--fragment--experimental).  This feature doesn't appear to be present in roslibjs.

Is there any plans to implement this? If not, is there some direction that can be provided in terms of getting this implemented?

For full context, I'm interested in using fragmentation in Foxglove, but [they explained](https://github.com/foxglove/studio/issues/4005) that they use roslibjs to handle this so such an update would have to occur here.

MatthijsBurgh, 2022-08-05:
I am open to supporting this feature.

Though I do not have the time to implement this. So when you want to implement this, I am willing to assist where I can. But the bulk should be done by you (or someone else).

EzraBrooks, 2025-11-03:
Done in #973!