RobotWebTools/roslibjs #558: Question about TfClient

roddc, 2022-08-08:
**Description**
I am using a TfClient to subscribe to tf2_web_republisher to get the transform feedback. However, the subscribe callback never called. If I subscribe to the TF inside a topic.subscribe, I can get the feedback, it seems that I can't not get the feedback outside the topic.subscribe. Does it mean I need to subscribe to the topic first before subscribing to the TF?

- Library Version: 
- ROS Version: Melodic
- Platform / OS: Ubuntu

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

MatthijsBurgh, 2022-08-08:
Please provide code samples

roddc, 2022-08-09:
> Please provide code samples

I have provided the code sample to reproduce, I was using a ros bag to test the code.

closed 2023-03-07 by roddc