RobotWebTools/roslibjs #560: TFClient unified goal_id

ufrhaidar, 2022-08-22:
if i was to use ros3djs to visualize a URDF or a 2d map, the `TFClient` sends a goal to the `tf2_web_republisher` with a `goal_id`. 

I would like to open the same visualization on multiple tabs or devices, and I don't want the `tf2_web_republisher` to send the same transforms multiple times only because the `goal_id` is different. Would it be possible to choose a constant goal_id to reduce the bandwidth of data being published by the `tf2_web_republisher`.

To get an idea of how bad the bandwidth gets, try the following [visualize a URDF tutorial](http://wiki.ros.org/ros3djs/Tutorials/VisualizingAURDF). Monitor the bandwidth on the feedback topic `rostopic bw tf2_web_republisher/feedback` while refreshing the URDF visualizer tab. 

Also, could you update the documentation of the `TFClient` to show how to cancel a specific goal? I've been using `this.TfClient.currentGoal.cancel();` to cancel a specific goal or `this.TfClient.actionClient.cancel();` to cancel all goals.

MatthijsBurgh, 2022-08-22:
> Also, could you update the documentation of the `TFClient` to show how to cancel a specific goal? I've been using `this.TfClient.currentGoal.cancel();` to cancel a specific goal or `this.TfClient.actionClient.cancel();` to cancel all goals.

Could you create a separate issue for this?

MatthijsBurgh, 2022-08-22:
I don't think this is possible. This would require a major refactor of the code. I don't have the resources to do that. So you would need to develop it mostly yourself. But when working correctly and not breaking current behaviour. I am happy to integrate it.

ufrhaidar, 2022-08-22:
I'm still new to javascript. Have tried to change the `goal_id` in `roslibjs/src/actionlib/Goal.js` and making it static. That didn't work as I might be somehow importing the wrong module. Any advise or comments are appreciated. If I get this simple change to work, I can get the `TFClient` to have an option of using a static `goal_id`.

MatthijsBurgh, 2022-08-23:
By changing the `goal_id` to be static. The `actionlib` becomes useless for normal action interfaces. So that is the difficulty of this. Which is why a major refactor would be required.
Also only one client should own a `goal_id`. So by using a static one. The whole canceling become very complex.

ufrhaidar, 2022-08-23:
yes I understand that. Making it static would only be a first step for me to test that the concept works. Later a different goal object can be created and used only with `TFClient` or `StaticTFClient`.

The solution would also require a custom or modified `tf2_web_republisher`. I already have the republisher made, and I'll keep working on the static goal_id when I have time. I'm just having trouble importing the newly built roslibjs.

MatthijsBurgh, 2022-08-23:
As long as you only changed code inside the functions, you should still be able to compile the library. Once you start adding classes. I think these should be in their own file. (Not 100% sure). And don't forget to extend the index files in that case.

EzraBrooks, 2025-11-08:
In my opinion, the correct way to implement this would not be to break the assumptions of the ROS protocol by allowing multiple clients to interact with a single goal handle, but instead to implement a SharedWorker interface that allows a Ros instance to be shared between tabs and windows. I believe that is doable in application code that wraps roslibjs instead of doing it in roslibjs itself.