RobotWebTools/roslibjs #563: Heartbeat for rosbridge

ravijo, 2022-09-08:
This question has been asked previously in [ROS Answers](https://answers.ros.org/question/334714/ros-heartbeat-for-rosbridge).

We have seen that the web client stays quiet even if the connection with the rosbridge server breaks (such as WiFi disconnects). Therefore, it is essential to have a built-in feature, like a heartbeat topic, that a rosbridge client can use to detect the connection status.

MatthijsBurgh, 2022-09-08:
The ROS class emits a close event when the connection get disconnected. You can use that.
I use that in my auto-ros wrapper, https://github.com/tue-robotics/auto-ros

ravijo, 2022-09-08:
Oh yeah. I completely forgot about the `close` event. It is there in my web app but just doing a `console.log()`, nothing else. Thanks for sharing the [auto-ros](https://github.com/tue-robotics/auto-ros) wrapper.

closed 2022-09-08 by ravijo