RobotWebTools/roslibjs #564: debug msg from _joint_jog.py

AltinayGrass, 2022-09-08:
**Description**
I am using node-red-contrib-flowake-ros-nodes to send servo_jog command. msg Type at below.
It is working quiet well but I am receiving below debug message around 2 times on every 10 messages. 

Any ideas to eliminate this error or bug?

- Library Version:1.3.0 
- ROS Version: humble
- Platform / OS: amd64 / ubuntu

**control_msgs/msg/Joint_Jog**

std_msgs/Header header
string[] joint_names
float64[] displacements
float64[] velocities
float64 duration

**Expected Behavior**

[image]

**Actual Behavior**

`ERROR] [1662661181.714759952] [rosbridge_websocket]: [Client b6ab2456-8942-415a-b079-a519c5c88be9] [id: publish:/servo_node/delta_joint_cmds:71] publish: The 'velocities' field must be a set or sequence and each value of type 'float' and each double in [-179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000, 179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000]
`

MatthijsBurgh, 2022-09-08:
I don't know in which way ROS2 is supported. I hope @RobotWebTools/ros2-0 can help you.

MatthijsBurgh, 2023-02-05:
Related to ro

closed 2023-02-05 by MatthijsBurgh