RobotWebTools/roslibjs #567: Question about client genertated by Service donnot disconnected when node exit

byebaibai, 2022-10-05:
**Description**

- Library Version: 1.3.0
- ROS Version: Foxy
- Platform / OS: Ubuntu 20.04.5 LTS

**Steps To Reproduce**

```
setInterval(function () {
  var statusRos = new ROSLIB.Ros();
  statusRos.connect(that.$rosBackendUrl);
  statusRos.getNodes(async function (nodes) {
    for(var ni in nodes) {
      const device_status_client = new ROSLIB.Service({
        ros: statusRos,
        name: "foo/get_device_status",
        serviceType: 'std_srvs/Empty'
      })
      async function getStatus() {
        return new Promise((resolve, reject) => {
          device_status_client.callService(
            new ROSLIB.ServiceRequest({}), 
            device => {
              // Do something
              resolve();
            },
            error => {
              reject(error);
              console.error(error);
            }
          );
        })
      }
      try {
        await getStatus();
      } catch( err ){
        console.error(err);
      }
    }
  }
  await statusRos.close();
}, 1000);
```

**Expected Behavior**
When using `kill -2 {NODE_PID}` to exit one node, the client caused by `callService` should **be disconnected** to the rosbridge_websocket, and the number of clients connected to the rosbridge_websocket should be minus one.

**Actual Behavior**
When using `kill -2 {NODE_PID}` to exit one node, the client caused by `callService` seems **not to be disconnected** to the rosbridge_websocket, and the number of clients connected to the rosbridge_websocket should hold the same.

MatthijsBurgh, 2022-10-05:
@byebaibai could you create a running example?

MatthijsBurgh, 2022-10-05:
The ROS object is disconnecting. Not the topics/service. So something else is going wrong.

closed 2023-02-05 by MatthijsBurgh