RobotWebTools/roslibjs #586: Better and more documentation

snorresovold, 2023-01-04:
The documentation for Roslibjs is very poor and there is far from enough.

the example folder has a few examples that work when you understand them, but there isn't enough and it's all written in plain html.

We need more examples for doing more stuff so people have the ability to learn by example and see how other people may have solved problems they have.

we also need documentation in other versions than plain html, I for example use React and Typescript, I had a very hard time making Roslibjs work  in this environment even though it's an npm package with at the time of writing this almost 3 thousand weekly downloads.

MatthijsBurgh, 2023-01-04:
Sorry, but I don't have the time do this. I am happy to assist people. It would be nice if questions of people would lead to more docs and examples. But this is an open-source project. So it requires a community to keep it up.

Unless you are going to pay me money to work on this full-time.... But you are probably not going to do that.

But if I understand correctly, you now have a React example. please share it with us. It gives you credit, so people will help you in the future too.

snorresovold, 2023-01-04:
Yes i was thinking of maybe adding it sometime after lunch, how is the process of cotributing to the project?

MatthijsBurgh, 2023-01-04:
Just fork the repo, commit your changes to your fork and open a PR. Please don't commit your build files, CI will update them.

snorresovold, 2023-01-04:
Made a simple example just to test if everything is correct. Does this look good to you @MatthijsBurgh ? #587

snorresovold, 2023-01-04:
If this looks correct then I'll add a few more examples today, but I don't know how many, I'll probably then continue to add some as i find them

snorresovold, 2023-01-04:
Just found this project as well. https://github.com/pantor/ros-control-center
This can be used as e good example for using angular.

NaiveInvestigator, 2024-04-03:
hello, newbie here, when trying to run the example [simple.html](https://github.com/RobotWebTools/roslibjs/blob/develop/examples/simple.html). i get this error:

```
Loading failed for the <script> with source “file:///home/maelstrom/repos/roslibjs/dist/RosLib.umd.cjs”. simple.html:6:38
Uncaught ReferenceError: ROSLIB is not defined
    <anonymous> file:///home/maelstrom/repos/roslibjs/examples/simple.html:11
simple.html:11:13
    <anonymous> file:///home/maelstrom/repos/roslibjs/examples/simple.html:11
``` 

I'm not quite sure what is the issue

EzraBrooks, 2025-11-03:
#597 added a React example. Other than that, this issue is too vague to be addressed - I think the JSDoc comments in this repository are great places to add (non-framework-specific) example usage notes!