RobotWebTools/roslibjs #599: How to use the Move Group Interface with roslibjs (RobotWebTools)?

laibazahid26, 2023-03-07:
I created a python node in which I am generating a trajectory for a Robotic Arm ([using python move group interface](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html)) for going from point A to point B. I generated this trajectory via:

    (plan, fraction) = move_group.compute_cartesian_path(waypoints, 0.01, 0.0 ) 

 *waypoints is a list of coordinates for the robot end effector to go.*

I saved this 'plan', returned from the above code, as a text file. 

Now I want to execute this plan using roslibjs. I cannot find any syntax help in roslibjs documentation GitHub. 

I did half of the thing in the python ros node (the generation of trajectory) because I couldn't find how I can do this via roslibjs. 

Is there any way of using the move the group interface in roslibjs?

MatthijsBurgh, 2023-03-07:
There is no specific class to interact with the move group ROS interface and provide some extra functionality. You should just the "raw" topics/services/action interfaces.

laibazahid26, 2023-03-07:
Any ideas on how I can go about this problem? How can I publish a 'plan' ( in the form of a .txt file) on the topic responsible for executing the trajectory via roslibjs?  The plan that is generated by the move group interface has a very cumbersome format. It is essentially an array of thousands of numbers. How can I input this in a 'message' format in roslibjs?

MatthijsBurgh, 2023-03-07:
That sound like an issue the internet should have figured out. You should probably save the data as a json/yaml. And then i guess there are many solutions to get them into an JS object. But that is outside the scope of this library and so my support.

But why do you save the plan from python? Why not execute it there as well?

laibazahid26, 2023-03-07:
The trajectory needs to be executed when a button is clicked on the browser roslibjs GUI.

MatthijsBurgh, 2023-03-07:
Maybe easier to create a topic, could be even empty msg, from the web interface to the python app. Which then executes the plan.

laibazahid26, 2023-03-07:
For this, the python node needs to keep running. Trying to make the python node and the GUI independent of each other. Idea was to keep the python node just for trajectory planning.

closed 2023-03-07 by laibazahid26