RobotWebTools/roslibjs #603: Unable to access http://robotwebtools.org/roslibjs

MichellePisciotta, 2023-04-07:
**Description**
Dear Author, I am writing to inform you that I am unable to access the website http://robotwebtools.org/roslibjs. Whenever I try to access the website, I receive an error message that says "HTTP ERROR 502". I have tried accessing the website from multiple devices and networks, but the issue persists. I am not sure if this is a temporary issue or if there is a problem with the website. I would appreciate it if you could look into this issue and provide an update on when the website will be accessible again. I rely on roslibjs for my work, and the inability to access the website is causing significant inconvenience. Thank you for your attention to this matter. Best regards,

MatthijsBurgh, 2023-04-07:
This is just seems to be an issue with GitHub. The entire website http://robotwebtools.org is not reachable. Which is just a custom domain to http://robotwebtools.github.io. Which is also not working.

So you just have to be patient.

MichellePisciotta, 2023-04-07:
> This is just seems to be an issue with GitHub. The entire website http://robotwebtools.org is not reachable. Which is just a custom domain to http://robotwebtools.github.io. Which is also not working.
> 
> So you just have to be patient.
Thank you so much for the response

MatthijsBurgh, 2023-04-10:
Discussion in https://github.com/RobotWebTools/robotwebtools.github.io/issues/82

closed 2023-04-10 by MatthijsBurgh