RobotWebTools/roslibjs #614: Correct CDN to use

torydebra, 2023-05-15:
**Description**

- Library Version: last
- ROS Version: noetic
- Platform / OS: ubuntu 20

**Steps To Reproduce**
The CDN indicated in the [ROS tutorials](http://wiki.ros.org/roslibjs/Tutorials/ActionlibClient) seems to not work anymore : 
```html
<script src="http://static.robotwebtools.org/EventEmitter2/current/eventemitter2.js"></script>
<script src="http://static.robotwebtools.org/roslibjs/current/roslib.js"></script>
```

Is the new correct CDN to be used the one indicated in the readme? What about the `eventemitter2.js` file?
```
https://cdn.jsdelivr.net/npm/roslib@1/build/roslib.min.js
```

Thanks!

MatthijsBurgh, 2023-05-15:
Forget about the ROS wiki, that one is really out-of-date.

The README is correct.

For `eventemitter2` use `https://cdn.jsdelivr.net/npm/eventemitter2@6/lib/eventemitter2.min.js` I will add this to the README.

MatthijsBurgh, 2023-05-15:
I think EventEmitter2 is included in the build and therefore no EventEmitter CDN is needed.

I will run the examples later to confirm this.

MatthijsBurgh, 2023-05-17:
Yes, I have tested the examples without EventEmitter CDN and these work.