RobotWebTools/roslibjs #646: What to do about ROS 2?

sea-bass, 2023-11-07:
I have recently implemented ROS 2 action support in `rosbridge_suite` in https://github.com/RobotWebTools/rosbridge_suite/pull/886, and a working client in `roslibjs` in https://github.com/RobotWebTools/roslibjs/pull/645.

However, this leaves a few questions open regarding ROS 2 support for this library. Namely,
* Right now, there is still a lot of functionality not ported to ROS 2 (with the exception of topics/services, basically)
* There are no ROS 2 examples
* CI is only running on melodic/noetic, but there are no jobs for e.g. humble/iron

So, what should the plan be going forward?
* Keep the `develop` branch on ROS 1 and start tackling the CI/testing revamping on the `ros2` branch?
* Spinning up a new `ros2libjs` repo?
* Other?

tapanij, 2023-12-05:
Hello @sea-bass !

I'm just getting to know ROS. I was wondering when a version of roslibjs and rosbridge_suite will be available in the rolling distribution that would allow this example action to be sent from roslibjs through rosbridge to the simulation.

[https://github.com/cyberbotics/webots_ros2/wiki/Example-Universal-Robots](https://github.com/cyberbotics/webots_ros2/wiki/Example-Universal-Robots)

```
ros2 action send_goal /ur5e/ur_joint_trajectory_controller/follow_joint_trajectory control_msgs/action/FollowJointTrajectory "{
  trajectory: {
    joint_names: [shoulder_pan_joint, shoulder_lift_joint, elbow_joint, wrist_1_joint, wrist_2_joint, wrist_3_joint],
    points: [
      { positions: [3.02, -1.63, -1.88, 1.01, 1.51, 1.13], time_from_start: { sec: 5, nanosec: 500 } },
      { positions: [-1.01, 0.38, -0.63, -0.88, 0.25, -1.63], time_from_start: { sec: 6, nanosec: 500 } },
      { positions: [-1.01, 0.38, -0.63, -0.88, 0.25, 6.2], time_from_start: { sec: 50, nanosec: 500 } }
    ]
  },
  goal_tolerance: [
    { name: shoulder_pan_joint, position: 0.01 },
    { name: shoulder_lift_joint, position: 0.01 },
    { name: elbow_joint, position: 0.01 },
    { name: wrist_1_joint, position: 0.01 },
    { name: wrist_2_joint, position: 0.01 },
    { name: wrist_3_joint, position: 0.01 }
  ]
}"
```

Is there a timeline known or can you give an estimate? Thank you very much!

sea-bass, 2023-12-05:
You can use this PR branch right now, actually!

https://github.com/RobotWebTools/roslibjs/pull/645

tapanij, 2023-12-07:
But the last rosbridge_suite release ([1.3.2](https://github.com/RobotWebTools/rosbridge_suite/tags)) is from [September ](https://github.com/RobotWebTools/rosbridge_suite/pull/886)and therefore doesn't include the action update?
The example will work with the upcoming rosbridge_suite release? Any idea when that will happen? Thanks!

tapanij, 2023-12-13:
@sea-bass 
I managed to build rosbridge_suite with the latest ros2-actions commit but the problem didn't change. I wrote a detailed question to robotics stack exchange, can you take a look at it please: 
[Issues with Sending ROS2 Action from roslib.js to rosbridge_suite: Unable to Import Message Classes](https://robotics.stackexchange.com/questions/107035/issues-with-sending-ros2-action-from-roslib-js-to-rosbridge-suite-unable-to-imp)
Thank you!

sea-bass, 2023-12-13:
The approach you described in that stackexchange question will install `rosbridge_suite` correctly, but to get `roslibjs` on this branch you can take one of two approaches:

1. Install this from the GitHub branch: https://www.warp.dev/terminus/npm-install-from-github
  * `pnpm install roslib@github:RobotWebTools/roslibjs#ros2-actions`

2. Clone this branch and link it: https://pnpm.io/cli/link
  * `cd /path/to/some/folder
  * `git clone -b ros2-actions git@github.com:RobotWebTools/roslibjs.git`
  * `pnpm link /path/to/some/folder/roslibjs`

tapanij, 2023-12-14:
@sea-bass That's good to know; I had no idea NPM could do that. I made the changes and installed ros2-actions with the first command. However, it didn't fix the problem, and I get the same errors on Rosbridge. So does that mean the problem is somewhere else than in roslib.js?

sea-bass, 2023-12-14:
Oh I just saw your stack exchange issue. I think you are using the ROS 1 compatible version of the action client, not the ROS 2.

Try following the example here: https://github.com/RobotWebTools/roslibjs/blob/ros2-actions/examples/ros2_action_client.html

tapanij, 2023-12-14:
Thanks, I got it working! I actually looked at the example, but I didn't notice the different classes for ros2.

sea-bass, 2023-12-14:
Yeah, the ROS 2 Implementation is totally different from the ROS 1, and this is admittedly very confusing.

We're working on cutting a v2 version of this library, stripping away the ROS 1 support, and keeping the v1 as the "final" ROS 1 release.

fllay, 2024-03-16:
I tried to port "tf2_web_republisher" to ros2. I can get action for tf woking. I tested with action CLI. However, when I am using roslibjs I got this error

```
[rosbridge_websocket-1] [INFO] [1710521016.292449524] [rosbridge_websocket]: Client disconnected. 0 clients total.
[rosbridge_websocket-1] [INFO] [1710521016.425017275] [rosbridge_websocket]: Calling services in existing thread
[rosbridge_websocket-1] [INFO] [1710521016.432058851] [rosbridge_websocket]: Sending action goals in existing thread
[rosbridge_websocket-1] [INFO] [1710521016.440076698] [rosbridge_websocket]: Client connected. 1 clients total.
[rosbridge_websocket-1] [INFO] [1710521016.459882491] [rosbridge_websocket]: [Client 2e0eb2d0-a6e6-4d53-9a9a-336c920f2697] Subscribed to /chatter
[rosbridge_websocket-1] [ERROR] [1710521016.470028692] [rosbridge_websocket]: [Client 2e0eb2d0-a6e6-4d53-9a9a-336c920f2697] [id: advertise:tf2_web_republisher/goal:2] advertise: tf2_web_republisherGoal is not a valid type string
[rosbridge_websocket-1] [INFO] [1710521016.495058242] [rosbridge_websocket]: [Client 2e0eb2d0-a6e6-4d53-9a9a-336c920f2697] Subscribed to tf2_web_republisher/status
[rosbridge_websocket-1] [ERROR] [1710521016.503971063] [rosbridge_websocket]: [Client 2e0eb2d0-a6e6-4d53-9a9a-336c920f2697] [id: subscribe:tf2_web_republisher/feedback:5] subscribe: tf2_web_republisherFeedback is not a valid type string
[rosbridge_websocket-1] [ERROR] [1710521016.510074906] [rosbridge_websocket]: [Client 2e0eb2d0-a6e6-4d53-9a9a-336c920f2697] [id: subscribe:tf2_web_republisher/result:6] subscribe: tf2_web_republisherResult is not a valid type string
```
I build `rosbridge_suite` using ros2 branch and used roslibjs v2 branch.  The javascript to cal the service is 

```
        var fibonacciClient = new ROSLIB.ActionClient({
            ros: ros,
            serverName: "tf2_web_republisher",
            actionName: "tf2_web_republisher",
          });

          // Create a goal.
          var goal = new ROSLIB.Goal({
            actionClient: fibonacciClient,
            goalMessage: {
              trans_thres: 1,
              angular_thres: 0.1,
              rate: 10.0,
              target_frame: "map",
              source_frames: ['map', 'map']
            },
          });
```

When I use this parameter to call the action in CLI, It works just fine. Any suggestions? Thank you.

sea-bass, 2024-03-16:
Looking at your example, I think `tf2_web_republisher` shouldnt be the value of the `actionName` argument. I think it should be `tf2_web_republisher/TFSubscription`, per:  https://github.com/RobotWebTools/tf2_web_republisher/blob/develop/action/TFSubscription.action

I think @pac48 is close to having a working prototype on both the tf2_web_republisher and roslibjs side. Just letting you know, as his fork has changed the message package to be more "correct" to how ROS 2 does things, with separate message packages:  https://github.com/pac48/tf2_web_republisher

fllay, 2024-03-16:
Thank you for letting me know about @pac48 work.  I think I will wait for him to finish the working prototype for both roslibjs and tf2_web_republisher.

EzraBrooks, 2024-06-28:
I think for now, the decision here is that roslibjs 2.0 will be adding some ROS 2 specific functionality but not regressing ROS 1. We should try our best to continue that strategy, but there may be another decision point later (hopefully after ROS 1 is officially EOL)

closed 2024-06-28 by EzraBrooks