RobotWebTools/roslibjs #647: Increase general library quality

MatthijsBurgh, 2023-11-16:
The [robostack fork](https://github.com/RoboStack/roslibjs), has applied some improvements like:
- Migrating to typescript
- Migrating to ES6 classes
- Replacing grunt by rollup

This might be good improvements to apply to the original library as well. Preferably we include these changes also in the `ros3djs` and `ros2djs` libraries.

EzraBrooks, 2023-11-16:
Agreed. I would recommend we start by adding a TypeScript compiler configuration that allows JavaScript files, fixing any issues with statically type checking the current JS, then move forward with TS.

EzraBrooks, 2024-01-24:
Once #665 (and all the dependent PRs that have come out of it) are in, I plan on making a PR to switch from Grunt + Browserify + Karma to Vite (which includes Rollup) and Vitest, as that's what I have experience with using lately and I think Vite's "library mode" is a very sensible preset for a package like roslibjs.

PR #687 (2024-02-21): Remove all remaining Grunt tasks in favor of Vite configs
**Public API Changes**

Should be none. Module resolution changes should all have been captured in #665 and the package.json config changes in this PR.

**Description**

- Remove browserify/uglify in favor of Vite in library mode.
- Replace grunt-jsdoc with an npm script
- Replace grunt-eslint with vite-plugin-checker
- Replace invocations of `tsc` with vite-plugin-checker
- Use dynamic imports for browser/node.js environment differences
- Remove various Grunt plugins

Closes #647.