RobotWebTools/roslibjs #660: V2 releases

MatthijsBurgh, 2023-12-19:
# Release procedure
- Provide build files?  Attached to the release?

# ToDo before first release
- Fix how examples are run
- Decide on what to do with eventemitter2 (https://github.com/RobotWebTools/roslibjs/pull/344)
- Remove or fix `socket.io` (https://github.com/RobotWebTools/roslibjs/pull/648), I think this one should be solved before the first v2 release

# ToDo along the way
- https://github.com/RobotWebTools/roslibjs/pull/647, I think this can be implemented along the way.

blackhorsewu, 2024-01-19:
> The roslibjs release on NPM still works. We're working on releasing 2.0, which will include new features. I'm not sure about ros3djs.

Thanks very much.

blackhorsewu, 2024-01-19:
I have tried `npm install roslibjs`, and it reported that "No valid versions available for roslibjs". Then I tried `npm install roslib`, it warned me "no such file or directory, open '/home/victor/flask_apps/package.json". I ignored it and run the `urdf.html` example. Then flask reported "GET /build/roslib.js HTTP/1.1" 404 -. So, where can I find /build/roslib.js and where should I put it?

blackhorsewu, 2024-01-19:
@EzraBrooks Thank you for your help and the examples. I have tried your "urdf.html". There I used <script src="https://cdn.jsdelivr.net/npm/roslib@1/build/roslib.js"></script> to get the "roslib.js" and now the web page is working and showed "Connected" in Green.
My next question is how do I display the urdf model on the web page? I have seen there is also a "urdf.html" example in ros3djs. However, I do not know how to feed the `urdfModel` of your "urdf.html" from my ROS into the `ROS3D.UrdfClient`. Hopefully the "urdf.html" example of ros3djs can visualize my urdf model. Thank you for your help.

EzraBrooks, 2024-03-20:
@MatthijsBurgh Do you think we're good to go to release 2.0 now? I've been using recent builds of the library on my project for quite awhile now and it seems very stable.

I think that instead of attaching build outputs to the release on GitHub we can probably just point users to the UMD version of the build output via [unpkg](https://unpkg.com) so we don't have to duplicate effort between publishing to NPM and publishing a CDN build.

EzraBrooks, 2024-03-20:
@blackhorsewu Sorry I missed your comment here. I actually don't have much experience with ros3djs. I personally use https://github.com/gkjohnson/urdf-loaders. However, you're welcome to open a Discussion if you're still looking for advice.

EzraBrooks, 2024-03-21:
that makes sense. we use ros2 on my project so it definitely works!

tall1, 2024-03-26:
> that makes sense. we use ros2 on my project so it definitely works!

Can you please provide a guide on how to run roslibjs on ros2?

YannickdeHoop, 2024-06-19:
@MatthijsBurgh Any updates on a release for ros2?

EzraBrooks, 2024-06-19:
I haven't had time to add some ROS 2 integration tests - I'd be happy to review a PR that introduces some so we can check that requirement off and do a release!

MatthijsBurgh, 2024-06-19:
Sorry, but I am in the same situation as @EzraBrooks

EzraBrooks, 2024-06-28:
Opened a dedicated issue to track ROS 2 testing #746

closed 2024-06-28 by EzraBrooks