RobotWebTools/roslibjs #714: Develop branch doesn't work with npm - React

RamiCMT, 2024-04-16:
**Description**
React: NPM doesn't package roslib properly when using "npm install" after specifying latest 'develop' commit in package.json

- Library Version: develop branch - latest commit
- ROS Version: Iron
- Platform / OS: Ubuntu 22.04 LTS

**Steps To Reproduce**

I'm developing a web interface using roslibjs within React. I decided to use the "develop" branch for ROS2 support, as the "ros2" branch hasn't been updated in years. To do this, I specified the commit SHA in the package.json file, using the following syntax and command:

"roslib": "github:RobotWebTools/roslibjs#253ae642782bddb3f942b19dc6462007d9f249c7",
$ npm install

I had been using an older commit (the one shown above) as **the most recent "develop" commit doesn't seem to package roslib properly with NPM;** the pictures below of the VS Code file explorer should highlight the issue. Due to this issue, I was unable to import roslib functionality and had to revert to the older commit.

That older commit had been working fine, but there was an issue with how the lib was using "Websocket" in the node_modules/roslib/src/core/Ros.js file. If I attempted to reconnect to the rosbridge using the same "Ros" object, the "WebSocket.CLOSED" used in the comparison on line 76 causes the following issue:
file:///home/ramiqamhieh/Pictures/Screenshots/Screenshot%20from%202024-04-16%2015-25-27.png
This WebSocket issue is irrelevant; this is simply the reason I decided to try to switch to the most recent commit.

**Expected Behavior**
From 'develop' commit I had been using (253ae642782bddb3f942b19dc6462007d9f249c7)
/node_modules/roslib
[image]

**Actual Behavior**
From latest 'develop' commit (cf1f5ada69e9f3f8e22887ffb000b0ca2dfc2277)
/node_modules/roslib
[image]

Full package.json file:
{
  "name": "ros2bridge",
  "version": "0.1.0",
  "private": true,
  "dependencies": {
    "@testing-library/jest-dom": "^5.17.0",
    "@testing-library/react": "^13.4.0",
    "@testing-library/user-event": "^13.5.0",
    "ol": "^8.2.0",
    "react": "^18.2.0",
    "react-dom": "^18.2.0",
    "react-router-dom": "^6.22.3",
    "react-scripts": "^5.0.1",
    "roslib": "github:RobotWebTools/roslibjs#cf1f5ada69e9f3f8e22887ffb000b0ca2dfc2277",
    "styled-components": "^6.1.8",
    "web-vitals": "^2.1.4"
  },
  "scripts": {
    "start": "react-scripts start",
    "build": "react-scripts build",
    "test": "react-scripts test",
    "eject": "react-scripts eject"
  },
  "eslintConfig": {
    "extends": [
      "react-app"
    ]
  },
  "browserslist": {
    "production": [
      ">0.2%",
      "not dead",
      "not op_mini all"
    ],
    "development": [
      "last 1 chrome version",
      "last 1 firefox version",
      "last 1 safari version"
    ]
  }
}

MatthijsBurgh, 2024-04-17:
@EzraBrooks could you have a look at this? As you contributed most changes, except from dependabot, https://github.com/RobotWebTools/roslibjs/compare/253ae642782bddb3f942b19dc6462007d9f249c7...cf1f5ada69e9f3f8e22887ffb000b0ca2dfc2277

EzraBrooks, 2024-04-17:
Sure. I have been using it with React via NPM but have been publishing a pre-release version of the package to my own registry, not referencing it directly via Git. It should be possible to use some lifecycle hooks to make it work properly with this workflow.

EzraBrooks, 2024-04-17:
One issue here appears to be that the `publish` lifecycle hook is being called (at least when I install with PNPM, which is what I'm using on my current project), and `publish` runs the tests in this repo.

Looking at the [NPM documentation](https://docs.npmjs.com/cli/v6/using-npm/scripts), this particular issue should be solved by moving testing into `prepublishOnly`, which is not invoked for local installations like `prepublish` and `publish` are.
https://github.com/RobotWebTools/roslibjs/blob/cf1f5ada69e9f3f8e22887ffb000b0ca2dfc2277/package.json#L52

PR #715 (2024-04-18): Split `publish` NPM hook into `prepublishOnly` and `prepare`
**Public API Changes**

None.

**Description**

Closes #714. Currently, installing this package via Git triggers some unwanted lifecycle hooks. This makes `npm` or `pnpm` `install roslib@github:RobotWebTools/roslibjs#develop` work.
closed 2024-04-18 by EzraBrooks