RobotWebTools/roslibjs #725: warning: possible EventEmitter memory leak detected. 11 listeners added. Use emitter.setMaxListeners() to increase limit.

chen-xiaoxing, 2024-05-07:
**Description**

When I subscribe, I get the data. After the data is processed, I do unsubscription in the method. This subscription is issued when the scroll bar scrolls on the page, but it shows a memory leak.

- Library Version: angular@8+roslib@1.1.0+node@14.21.3+ts@3.5.3
- ROS Version:1.1.0
- Platform / OS: windows11+vscode

**Steps To Reproduce**

let arr= [];
this.mapRosService.observeLandmark.subscribe(res => {
      arr= [];
      (res.markers).forEach(res1 => {
        let str = res1.pose.position;
        let obj = {
          x: str.x,
          y: str.y,
        }
        arr.push(obj);
      })
      this.mapRosService.observeLandmark.unsubscribe();
    })

**Expected Behavior**
Memory will not leak

**Actual Behavior**
Memory  leak

MatthijsBurgh, 2024-05-07:
This is incomplete as we need the message your are receiving.

Plus when it is a bug in EventEmitter, we can't fix it. So please describe what roslib is doing wrong. Please note your are using a significantly old library and node version. So whatever your issue might be, I suggesst you first test the latest 1.X version. with node 18, as node 16 is aleady EOL.

chen-xiaoxing, 2024-05-07:
Yes, my node and roslib versions are a bit old. The company has no plans to upgrade them at the moment. Let’s get back to this question.If you subscribe to the same roslib more than 11 times, an error will be reported.
I encapsulated the subscription to roslib in a method. This method is called when the mouse scrolls. When I continue to scroll the mouse (subscribe more than 11 times), a memory leak appears.The browser console error message points to the location where roslib is subscribed.

EzraBrooks, 2024-07-02:
This issue is describing a "possible" memory leak. It is safe to ignore this unless the number is unexpectedly high. It's gone in roslibjs v2 because v2 uses EventEmitter3 which I believe does not log this warning.

closed 2024-07-02 by EzraBrooks