RobotWebTools/roslibjs #726: Subscription for PointCloud2 is not working

apoorvcn47, 2024-05-08:
**Description**
I am trying subscribe to a Pointcloud2 topic but my subscription callback function never triggers. Similar styled subsciptions work for Image and MarkerArray topics.

////////////////////////////////////
let pc_topic = new ROSLIB.Topic(
{
    ros : ros,
    // name : '/camera/color/image_raw',
    // messageType : 'sensor_msgs/msg/Image',
    name : '/camera/depth_registered/points',
    messageType : 'sensor_msgs/msg/PointCloud2',
    // name : '/rviz/pallet1',
    // messageType : 'visualization_msgs/MarkerArray',
    // latch : true
});
// render_ob.create_pointcloud_sub(pc_topic);
pc_topic.subscribe(function(message) 
{
    console.log('message.width');
    console.log(message.width);
    console.log(message.height);
});

////////////////////////////////////

I was wondering if anyone has come across this.

- Library Version: v2
- ROS Version: Humble
- Platform / OS: Ubuntu 22.04

spxvszero, 2024-05-13:
try 'sensor_msgs/PointCloud2' instead of 'sensor_msgs/msg/PointCloud2'

apoorvcn47, 2024-05-13:
> try 'sensor_msgs/PointCloud2' instead of 'sensor_msgs/msg/PointCloud2'

I tried both. It doesn't make a difference.

MatthijsBurgh, 2024-05-13:
I suggest you start with the [simple example](https://github.com/RobotWebTools/roslibjs/blob/develop/examples/ros2_simple.html) and start adding changes, not all at the same, just as minimal as possible.

And you should indeed use `sensor_msgs/PointCloud2'`.