RobotWebTools/roslibjs #746: Add ROS 2 integration tests

EzraBrooks, 2024-06-28:
We'd like to release roslibjs v2 with a lot of improvements (asynchronous Services, ROS 2 Actions, ROS 2 TFClient), but we don't yet have any integration tests in CI for the ROS 2 support. Once this is done, we can release v2!

sauk2, 2024-06-29:
What do you think about using [`setup-ros`](https://github.com/ros-tooling/setup-ros) instead of the current approach followed for ROS 1?

EzraBrooks, 2024-07-01:
That makes sense to me

ganatrask, 2024-07-01:
Hey @EzraBrooks I'd like to work on this issue can you share more details on what things need to be added for tests or CI for ROS?

EzraBrooks, 2024-07-02:
The bulk of the work is to add ROS 2 versions of [`test/build.bash`](https://github.com/RobotWebTools/roslibjs/blob/f643c433a4b0719ae058f0e1078e17e02e1241a5/test/build.bash) and [`.github/workflows/main.yml`](https://github.com/RobotWebTools/roslibjs/blob/f643c433a4b0719ae058f0e1078e17e02e1241a5/.github/workflows/main.yml).

labbomb, 2024-12-17:
It's been going on for a while, and now it's stuck? When will this milestone be released

EzraBrooks, 2024-12-17:
As you can see from the milestone, this is the last remaining issue. Feel free to submit a PR!

FlorianLebecque, 2025-03-03:
I've been trying to implement the test for ROS2, I am still having issue with all TFClient related test. How should we proceed ?

EzraBrooks, 2025-03-04:
PR here: https://github.com/RobotWebTools/tf2_web_republisher/pull/35

(thanks Paul)

FlorianLebecque, 2025-03-05:
I've implemented most of the ROS2 CI. But whatever I try, I can't make most of the test works. Can someone take a look and maybe we can find what I am missing ?

https://github.com/FlorianLebecque/roslibjs/tree/develop-ros2