RobotWebTools/roslibjs #778: Develop does not build without lock-file

mhosmar-cpr, 2024-09-06:
### Description
When building a library as a dependency the lock file in this repo is not honored. This is expected behavior. Unfortunately with the current versions specified in the package.json this library will not build. Even though your CI builds since it is using the lock file. FYI libraries should not commit their lock file for this exact reason. https://github.com/yarnpkg/yarn/issues/838#issuecomment-253362537

### Build Error
Below is the build error
```bash
src/urdf/UrdfModel.js:35:7 - error TS2322: Type 'Element | null' is not assignable to type 'Element | undefined'.
  Type 'null' is not assignable to type 'Element | undefined'.

35       xmlDoc = parser.parseFromString(string, MIME_TYPE.XML_TEXT).documentElement;
         ~~~~~~

Found 1 error in src/urdf/UrdfModel.js:35
```
### Lock File Diff

<details>
<summary>Below is the diff between the committed lock file and one generated fresh.</summary>

```log
├ @xmldom/xmldom@0.9.0 -> @xmldom/xmldom@0.9.2
├ @testing-library/react@16.0.0 -> @testing-library/react@16.0.1
│ └ @babel/runtime@7.24.4 -> @babel/runtime@7.25.6
├ @types/node@22.5.0 -> @types/node@22.5.4
│ └ undici-types@6.19.6 -> undici-types@6.19.8
├ @types/ws@8.5.10 -> @types/ws@8.5.12
├ eslint@9.9.0 -> eslint@9.9.1
│ ├ @eslint/config-array@0.17.1 -> @eslint/config-array@0.18.0
│ │ └ debug@4.3.6 -> debug@4.3.7
│ │   └ ms@2.1.2 -> ms@2.1.3
│ ├ @eslint/js@9.9.0 -> @eslint/js@9.9.1
│ ├ esquery@1.5.0 -> esquery@1.6.0
│ ├ ignore@5.3.1 -> ignore@5.3.2
│ ├ optionator@0.9.3 -> optionator@0.9.4
│ │ ├ - @aashutoshrathi/word-wrap@1.2.6
│ │ └ + word-wrap@1.2.5
│ ├ fastq@1.16.0 -> fastq@1.17.1
│ └ acorn@8.12.0 -> acorn@8.12.1
├ jsdoc@4.0.2 -> jsdoc@4.0.3
│ ├ @babel/parser@7.23.9 -> @babel/parser@7.25.6
│ │ └ + @babel/types@7.25.6
│ │   ├ + @babel/helper-string-parser@7.24.8
│ │   ├ + @babel/helper-validator-identifier@7.24.7
│ │   └ + to-fast-properties@2.0.0
│ ├ @jsdoc/salty@0.2.7 -> @jsdoc/salty@0.2.8
│ ├ @types/markdown-it@12.2.3 -> @types/markdown-it@14.1.2
│ │ ├ @types/linkify-it@3.0.5 -> @types/linkify-it@5.0.0
│ │ └ @types/mdurl@1.0.5 -> @types/mdurl@2.0.0
│ ├ markdown-it@12.3.2 -> markdown-it@14.1.0
│ │ ├ entities@2.1.0 -> entities@4.5.0
│ │ ├ linkify-it@3.0.3 -> linkify-it@5.0.0
│ │ │ └ uc.micro@1.0.6 -> uc.micro@2.1.0
│ │ ├ mdurl@1.0.1 -> mdurl@2.0.0
│ │ └ + punycode.js@2.3.1
│ └ underscore@1.13.6 -> underscore@1.13.7
├ typescript@5.5.2 -> typescript@5.5.4
├ vite@5.4.0 -> vite@5.4.3
│ ├ postcss@8.4.41 -> postcss@8.4.45
│ │ └ picocolors@1.0.1 -> picocolors@1.1.0
│ └ rollup@4.13.0 -> rollup@4.21.2
├ vitest@2.0.2 -> vitest@2.0.5
│ ├ @vitest/expect@2.0.2 -> @vitest/expect@2.0.5
│ │ ├ @vitest/spy@2.0.2 -> @vitest/spy@2.0.5
│ │ └ @vitest/utils@2.0.2 -> @vitest/utils@2.0.5
│ │   └ @vitest/pretty-format@2.0.2 -> @vitest/pretty-format@2.0.5
│ ├ @vitest/runner@2.0.2 -> @vitest/runner@2.0.5
│ ├ @vitest/snapshot@2.0.2 -> @vitest/snapshot@2.0.5
│ ├ tinybench@2.8.0 -> tinybench@2.9.0
│ ├ tinypool@1.0.0 -> tinypool@1.0.1
│ ├ vite-node@2.0.2 -> vite-node@2.0.5
│ ├ why-is-node-running@2.2.2 -> why-is-node-running@2.3.0
│ └ npm-run-path@5.2.0 -> npm-run-path@5.3.0
├ @babel/code-frame@7.23.5 -> @babel/code-frame@7.24.7
│ ├ @babel/highlight@7.23.4 -> @babel/highlight@7.24.7
│ │ ├ @babel/helper-validator-identifier@7.22.20 -> @babel/helper-validator-identifier@7.24.7
│ │ └ + picocolors@1.1.0
│ ├ - chalk@2.4.2
│ │ ├ - ansi-styles@3.2.1
│ │ │ └ - color-convert@1.9.3
│ │ │   └ - color-name@1.1.3
│ │ ├ - escape-string-regexp@1.0.5
│ │ └ - supports-color@5.5.0
│ │   └ - has-flag@3.0.0
├ tiny-invariant@1.3.1 -> tiny-invariant@1.3.3
├ vscode-languageserver-textdocument@1.0.11 -> vscode-languageserver-textdocument@1.0.12
├ @volar/typescript@2.3.4 -> @volar/typescript@2.4.2
│ └ @volar/language-core@2.3.4 -> @volar/language-core@2.4.2
│   └ @volar/source-map@2.3.4 -> @volar/source-map@2.4.2
├ agent-base@7.1.0 -> agent-base@7.1.1
├ micromatch@4.0.5 -> micromatch@4.0.8
├ minimatch@3.0.8 -> minimatch@3.1.2
├ semver@7.5.4 -> semver@7.6.3
│ └ - lru-cache@6.0.0
│   └ - yallist@4.0.0
├ @volar/language-core@2.4.0-alpha.18 -> @volar/language-core@2.4.2
│ └ @volar/source-map@2.4.0-alpha.18 -> @volar/source-map@2.4.2
├ @vue/compiler-dom@3.4.19 -> @vue/compiler-dom@3.5.3
│ ├ @vue/compiler-core@3.4.19 -> @vue/compiler-core@3.5.3
│ │ └ @vue/shared@3.4.19 -> @vue/shared@3.5.3
├ pkg-types@1.1.3 -> pkg-types@1.2.0
├ @volar/typescript@2.4.0-alpha.18 -> @volar/typescript@2.4.2
├ binary-extensions@2.2.0 -> binary-extensions@2.3.0
├ is-core-module@2.15.0 -> is-core-module@2.15.1
└ ajv@8.17.1 -> ajv@8.13.0
  ├ - fast-uri@3.0.1
  └ + uri-js@4.4.1
    └ + punycode@2.3.1
```
</details>

### Solution

Ideally the package.json should be updated to reflect the required versions of dependencies.

MatthijsBurgh, 2024-09-09:
I don't think some minor/patch version updates do break the build. This should not be the case. So please figure out what package version requirement is incorrect.

The line of code which is causing you issues was updated in combination of a minor version bump of `@xmldom/xmldom`. So unless one of the bugfixes in the patch version upgrades are needed, That one can't be the issue. But other packages shouldn't be relevant.

So please do some research on which package is the issue for you.

mhosmar-cpr, 2024-09-16:
Have you tried building without the lock file? Remove the node modules folder and run `npm install --package-lock false`

For now we are sticking with 6284a5f till this is fixed.

MatthijsBurgh, 2024-09-25:
Removing the lock file and running `npm install` fails indeed.

I am not sure it it is a bad practice to build with a lock file. But lets skip that point here.

When checking the new generate lock file in combination with the build errors, you could also have noticed a patch bump of `@xmldom/xmldom`. Something I mentioned. But you didn't put any effort in to look at. I am willing to help. But this is open-source software. So some effort from your side is also required.

The error shown is introduced in `0.9.2`. (`0.9.1` is also broken, but for another reason.)

closed 2024-09-25 by MatthijsBurgh