RobotWebTools/roslibjs #813: Running ros2_action_client.html Fails to Connect to ROS2 Action Server

howking111, 2024-11-07:
**Description**

- Library Version:
- ROS Version: humble
- Platform / OS: Ubuntu22.04

**Steps To Reproduce**

Error Log:
`[id: send_action_goal:/fibonacci:1] Unknown operation: send_action_goal.  Allowed operations: ['call_service', 'advertise', 'unadvertise', 'publish', 'subscribe', 'unsubscribe', 'fragment', 'advertise_service', 'service_response', 'unadvertise_service']`

**Expected Behavior**

**Actual Behavior**

sea-bass, 2024-12-29:
I think you may need to upgrade your `rosbridge` version to be 2.0.0, as this capability was added in that release.

Fluidmechanik, 2025-03-19:
Hi, @sea-bass
How to use the RosLib.js or RosLib.umd.cjs that i have built in the html file? 
i got the errors:
1)Uncaught ReferenceError: Buffer is not defined
2)Source map error: Error: NetworkError when attempting to fetch resource.
Resource URL: https://unpkg.com/eventemitter3@latest/dist/eventemitter3.umd.min.js
Source Map URL: eventemitter3.umd.min.js.map