RobotWebTools/roslibjs #840: Action message type import issue

beastGrendel, 2025-01-28:
**Description**
I am trying to create a browser based GUI to interact with my ROS2-backbone. I have implemented the action client example ros2_action_client.html as part of a component in the Angular framework. I am creating a ROSLIB instance as an angular-service which I can refer to from my component. I then utilize it to create an ActionClient in the component's constructor. For this example I have switched to using the messages defined in the action tutorial, however my custom message type does not work either - I get the same errors. I have built and sourced my workspace before executing rosbridge. When my GUI connects to the server through its websocket I get the following errors:

`
[rosbridge_websocket-1] [ERROR 1738063669.859900810] [rosbridge_websocket]: [Client <...>] [id: advertise:/fibonacci/goal:1] advertise: Unable to import action_tutorials_interfaces.msg from package action_tutorials_interfaces. Caused by: No module named 'action_tutorials_interfaces.msg' (log() at /home/<...>/ros2_ws/install/rosbridge_library/local/lib/python3.10/dist-packages/rosbridge_library/protocol.py:396)
[rosbridge_websocket-1] [INFO 1738063669.868994961] [rosbridge_websocket]: [Client <...>] Subscribed to /fibonacci/status (log() at /home/<...>/ros2_ws/install/rosbridge_library/local/lib/python3.10/dist-packages/rosbridge_library/protocol.py:400)
[rosbridge_websocket-1] [ERROR 1738063669.869433153] [rosbridge_websocket]: [Client <...>] [id: subscribe:/fibonacci/feedback:4] subscribe: Unable to import action_tutorials_interfaces.msg from package action_tutorials_interfaces. Caused by: No module named 'action_tutorials_interfaces.msg' (log() at /home/<...>/ros2_ws/install/rosbridge_library/local/lib/python3.10/dist-packages/rosbridge_library/protocol.py:396)
[rosbridge_websocket-1] [ERROR 1738063669.869779237] [rosbridge_websocket]: [Client <...>] [id: subscribe:/fibonacci/result:5] subscribe: Unable to import action_tutorials_interfaces.msg from package action_tutorials_interfaces. Caused by: No module named 'action_tutorials_interfaces.msg' (log() at /home/<...>/ros2_ws/install/rosbridge_library/local/lib/python3.10/dist-packages/rosbridge_library/protocol.py:396)
`

Rebuilding the project cleanly and then sourcing before launching rosbridge does produce the same errors, nothing changes

- Library Version: up-to-date rosbridge_suite branch humble with standard roslibjs installation (via npm) and Angular19
- ROS Version: ROS2 Humble
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**

1) Create a simple Angular app with a service providing the ROS instance.
2) Create a simple component where the ROS instance is used to set up a ROS ActionClient, as well as a button.
2) Use a button click-event to trigger the action request. Implement the client functions as defined in the action client example within the function responding to the click event.
3) Launch the rosbridge server
4)  Serve the angular app (ng serve)

**Expected Behavior**
Info logging notifying me about successfully subscribing to and advertising the topics / action. Action request sent upon button press.

**Actual Behavior**
Immediately upon establishing connection between roslibjs and rosbridge, the errors pop up. Upon button push there is another error, that the action cannot be inferred because it has not been advertised yet.

If required I can provide a demo app with just the angular-service and the button component.

f-o-w-l, 2025-01-29:
Sounds like it's possible you're using the wrong `serviceType` when instantiating the `new ROSLIB.Service`. When migrating one of my projects from ROS1 to ROS2, we needed to add `/srv/` between the namespace and service message definition name.

For example, ROS1 service instantiation (using TypeScript, ignore the generic types in the <> if you're not):
```
const postProcessingService = new ROSLIB.Service<PostProcessingReq, PostProcessingRes>({
  ros: rosClient,
  name: 'post_processing',
  serviceType: 'custom_msgs/PostProcessingRequest'
})
```
Same instantiation for ROS2:
```
const postProcessingService = new ROSLIB.Service<PostProcessingReq, PostProcessingRes>({
  ros: ros2Client,
  name: 'post_processing',
  serviceType: 'custom_msgs/srv/PostProcessingRequest'
})
```
Maybe this is your issue? Providing your relevant `new ROSLIB.Service` code might help people diagnose.

beastGrendel, 2025-01-31:
I am currently isolating the relevant code from my code base so I can show it here. I am trying to reimplement everything connected to this issue from the ground up so it as clean as possible, and I have noticed that, while in the ROS2 action example there is the option to create a variable through 'var arbAction = new ROSLIB.Action({...})', that option is not available in my local install of roslibjs (with npm install roslibjs). Also, the files 'Action.js' and 'GoalStatus.js' are missing in the folder '/node_modules/roslib/src/core'. I had previously tried to utilize the action client used in the action_client.html example. But this is only applicable for ROS1 then?

beastGrendel, 2025-02-15:
Ok, for any angular beginners: Just leave the methods that execute the publishing of action goals and the like centralized in a service where you also instantiate the roslib instance (where you call private _ros = new ROSLIB.Ros({})) . Outsourcing the goal publishing into component methods by referring to the roslib instance instantiated within the service via a getter function and using that locally within components did not work for me.

closed 2025-02-15 by beastGrendel