RobotWebTools/roslibjs #852: getServiceResponseDetails seems to return incorrect type

ahoenerBE, 2025-02-27:
**Description**

- Library Version:
- ROS Version:
- Platform / OS:

**Steps To Reproduce**

`rosapi` defines `ServiceResponseDetails` as
```idl
# ServiceResponseDetails.srv
string type
---
TypeDef[] typedefs
```

```idl
# TypeDef.msg
string type
string[] fieldnames
string[] fieldtypes
int32[] fieldarraylen
string[] examples
string[] constnames
string[] constvalues
```

RosLibJS documents in its callback that the type is going to be an object that contains a key `typdefs` that's a list of strings.
```js
/**
   * @callback getServiceResponseDetailsCallback
   * @param {{typedefs: string[]}} result - The result object with the following params:
   */
```

This seems like a documentation error?

**Expected Behavior**

**Actual Behavior**

MatthijsBurgh, 2025-02-28:
This looks indeed like a documentation error. You will get the full message.

MatthijsBurgh, 2025-02-28:
Feel free to open a PR to fix the documentation